ImaginaryRadicals/ftc_app #3: Kinematics of particle launch

ajtrowell, 2016-11-18:
Need a utility to estimate the required launch speed to put a particle in the center vortex at a given distance. Launch angle could be provided as an argument, or provided as a CONSTANT in Hardware_iRads.

It would also be good to know the minimum and maximum useful launch speeds for a given launch angle, given the minimum height to clear the vortex, and the maximum legal height of a launched particle.

stephenberry, 2016-11-19:
I'm wrapping up this code, but someone will need to implement this code with the launch motors and the rest of the logic.

stephenberry, 2016-11-26:
I added a repository for the trajectory code I wrote in BluJ, it's called Trajectory.