ImaginaryRadicals/ftc_app #4: Add Exponential input gains for better driver control.

ajtrowell, 2016-11-18:
A utility for adding exponential gains would make driving more precise.
A possible place for this utility to live is as a `static` method in a Utility class. 
`scaledOutput = Utility.expo(input,gain)`
This utility can be used to scale our raw gamepad stick inputs before sending them to the motors.

Here's some explanation and motivation of the exponential gains. Note that the response doesn't HAVE to be strictly exponential, a polynomial approximation could work, as long as it is tunable.

- [Youtube: What is expo?](https://www.youtube.com/watch?v=ajJ3rJ1HN5Q&index=12&list=PLYsWjANuAm4oWuE5ZDX4TAiHBv2i0Xl4m)
- [Expo response curve.](https://static.rcgroups.net/forums/attachments/4/0/1/6/3/a671305-160-Expo.jpg)
- [A simple expo approximation.](https://www.rcgroups.com/forums/showpost.php?p=12960286&postcount=11)

fantasticMan1, 2016-11-20:
I added an opmode that has only the exponential drive.  For now, it only has the drive functionality, but I will add the forklift and servos soon.

ajtrowell, 2016-11-20:
Good idea Josh. Here's a couple of suggestions:

Instead of the calculation class, make a more generic 'Utility' class. Not only is the name probably more appropriate, but since we have to type it every time we use one of these functions, we will appreciate the shorter name.

There is a library function in Java to raise numbers to a power. If you want 10 cubed, use:
`Math.pow(10,3);`

If you'd like to see how [this expo approximate](https://www.rcgroups.com/forums/showpost.php?p=12960286&postcount=11) works...
Checkout this [spreadsheet.](https://docs.google.com/spreadsheets/d/1c2SQ6pQjzkkAPeidS_6URh4yEGORbZnQpQZtfPeO0Bg/edit?usp=sharing)
Within the spreadsheet, you can modify the gain values above Approx1 and Approx2 to see how it changes the response curve.  Useful values for gain range from 1, which is a flat response, no expo gain. A gain of 5 is on the high end, and has a stronger effect than cubic.  

You should try to implement this fancier method in the expo(input,gain) function.

Finally, go ahead and implement the function in our REAL iRads opMode.  Modify the value going from the gamepad to the nextMotorState around line 155.

ajtrowell, 2016-11-22:
@elfergo 
Will, Josh did some work on this, but I'd like you to wrap it up. Read what we've discussed and let us know if you have questions.

commit ba057db7 (2016-11-25): Implemented Expo in Controls, Closes #4

used Utility.expo to make drive increase by a power of 3. Helpful for better driver control.
closed 2016-11-25 by ajtrowell