ImaginaryRadicals/ftc_app #17: Autonomous: Rotate robot until its absolute heading is X degrees

ajtrowell, 2016-12-10:
This routine is a bit different from issues #16 or #15 because it this issue requires the robot to change its ABSOLUTE position, not its RELATIVE position.  In #16, the goal was to rotate X degrees relative to its current position.  In this issue, the goal is to rotate until the robot heading is X degrees relative to the Game Field x-axis.

To do this, the robot's sensors and navigation classes must be used to determine absolute position.
_Visual Navigation_ will only work properly while the target images are in view, but have already been shown to work in some fashion. 
Use the `visionUpdate()` method as an example. The relevant methods are:
- `visualNav.getHeading()`
- `visualNav.getTrackAge()`

_Encoder Navigation_ is untested, but has the potential to be more robust than visual navigation. It is setup to update itself whenever visual navigation information is available, and to used encoder navigation whenever the image targets are out of sight.
Use the `encoderUpdate()` method as an example. The relevant methods are:
- `encoderNav.getHeading()`
- `encoderNav.getAbsoluteLocalizationAge()`
- `encoderNav.getAbsolutePositionDistance_mm()`

This method could call the one created in issued #16, and will likely be a useful component of more advanced autonomous routines.

fantasticMan1, 2016-12-22:
Wrote code that performs this in theory.  Testing continues.

ajtrowell, 2016-12-22:
The code is coming along.  We just need to decide how to package it, probably in a AutoDrive.rotateAbsolute(angleDegrees) Class/Method.  There are a few other methods that we'll want to add to help, like isRotationComplete() and things like that.